epsilonrt/modbus-serial #1: ESP32C3 hang when Coil read 8n's coil.

kscccc, 2023-04-08:
Whole Modbus program hang when read 8 or 16 or 8*n coils using Read Coil Status (FC=01).
However it is normal when read 1~7,9~15,17~23,etc.
The loop is still running. Seem something overflow in the program.

2023/04/08 23:17:24  >>> 0E 01 00 00 00 08 3D 33 
2023/04/08 23:17:24  < 0E 01 01 2A D3 43 43 4F 52 52 55 50 54 20 48 45 41 50 3A 20 42 61 64 20 74 61 69 6C 20 61 74 20 30 78 33 66 63 39 31 37 34 37 2E 20 45 78 70 65 63 74 65 64 20 30 78 62 61 61 64 35 36 37 38 20 67 6F 74 20 30 78 62 61 61 64 35 36 66 38 0D 0A

epsilonrt, 2023-04-09:
This bug concerns modbus-arduino, I move it.

epsilonrt, 2023-04-09:
Could you update modbus-arduino to version 1.1.1 and verify that your problem is gone ?
THANKS.
https://github.com/epsilonrt/modbus-arduino/archive/refs/tags/v1.1.1.zip