epsilonrt/modbus-serial #2: [Help] Modbus RS485 RTU Server can be accessed by mbpoll but not by pymodus

srolf, 2023-07-08:
Hi all,

I'm trying to reproduce the modbus-serial "Switch" example on ESP32.

```
<snip>
// Used Pins
const int SwitchPin = 25;
const int TxenPin = -1; // -1 disables the feature, change that if you are using an RS485 driver, this pin would be connected to the DE and /RE pins of the driver.

const byte SlaveId = 13;
// Modbus Registers Offsets (0-9999)
const int SwitchIsts = 0;

#define MySerial Serial2 // define serial port used, Serial most of the time, or Serial1, Serial2 ... if available
const unsigned long Baudrate = 115200;

// ModbusSerial object
ModbusSerial mb (MySerial, SlaveId, TxenPin);

void setup() {
  
  // MySerial.begin (Baudrate); // works on all boards but the configuration is 8N1 which is incompatible with the MODBUS standard
  // prefer the line below instead if possible
  MySerial.begin (Baudrate, MB_PARITY_EVEN);
<snip>
```

But I'm stumbling to reproduce/complete it by using pymodbus on a Raspberry Pi2.

On the ESP32 Serial2 is used and a RS485 TTL adapter is connected. It's working by just using a serial communication.
Also on the RPi using mbpoll works and reports the button status (pressed/released) correctly:

```
$ mbpoll -m rtu -b115200 -P even -a13 -t1 /dev/serial0
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU
Slave configuration...: address = [13]
                        start reference = 1, count = 1
Communication.........: /dev/serial0,     115200-8E1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: discrete input

-- Polling slave 13... Ctrl-C to stop)
[1]: 	0
-- Polling slave 13... Ctrl-C to stop)
[1]: 	0
-- Polling slave 13... Ctrl-C to stop)
[1]: 	1
-- Polling slave 13... Ctrl-C to stop)
[1]: 	1
-- Polling slave 13... Ctrl-C to stop)
[1]: 	0
-- Polling slave 13... Ctrl-C to stop)
[1]: 	0
^C--- /dev/serial0 poll statistics ---
6 frames transmitted, 6 received, 0 errors, 0.0% frame loss
```
But by trying to write a modbus client with pymodbus, I got stuck:

```
import pymodbus
import serial
from pymodbus.pdu import ModbusRequest
from pymodbus.client import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer

client= ModbusClient(method = "rtu", port="/dev/serial0",stopbits = 1, bytesize = 8, parity = 'E', baudrate= 115200)

#Connect to the serial modbus server
connection = client.connect()
print (connection)

#Starting add, num of reg to read, slave unit.
result= client.read_input_registers(address=13, count=1)

print(result)

#Closes the underlying socket connection
client.close()
```
The output looks like this:

```
True
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
```
Any idea what I'm doing wrong?

epsilonrt, 2023-07-09:
Hi,
If you have an answer by mbpoll and not by pymodbus it is that the problem comes from your use of pymodbus.
I suggest you look into pymodbus timeout setting, to do unit testing you can use [pymodbus.console](https://pymodbus.readthedocs.io/en/latest/source/library/REPL.html ). You can also use an oscilloscope or a logic analyzer or [ssniffer](https://github.com/epsilonrt/tty0tty/tree/master/ssniffer) to spy on the serial link.
In any case, this does not concern Modbus-Serial, post a message on the pymodbus forum.

closed 2023-07-09 by epsilonrt

srolf, 2023-07-09:
Thank you for the hint. I will have a look at ssniffer, maybe I get an idea what's going wrong. The timeout setting might be also a point to start. It's my first time using modbus at all... so I'm still uncertain at any step :-)